mledan/OctoPrint-EasyServo #6: Installation fail

VASAPOL, 2020-05-23:
ERROR: HTTP error 404 while getting https://github.com/iFrostizz/EasyServo/archive/master.zip
  ERROR: Could not install requirement https://github.com/iFrostizz/EasyServo/archive/master.zip because of error 404 Client Error: Not Found for url: https://github.com/iFrostizz/EasyServo/archive/master.zip
ERROR: Could not install requirement https://github.com/iFrostizz/EasyServo/archive/master.zip because of HTTP error 404 Client Error: Not Found for url: https://github.com/iFrostizz/EasyServo/archive/master.zip for URL https://github.com/iFrostizz/EasyServo/archive/master.zip

M0o0P, 2020-05-23:
The URL is incorrect as it doesn't include "Octoprint-" use this URL to install:

https://github.com/iFrostizz/OctoPrint-EasyServo/archive/master.zip

@iFrostizz, Please update here and on Octoprint. Thank you for putting this plugin out, I've been wanting to write this for a year now.

iFrostizz, 2020-05-24:
Thanks, it should be good now

closed 2020-05-24 by iFrostizz