mledan/OctoPrint-EasyServo #7: Install instructions incorrect for pigpio lib install

mcstoufer, 2020-06-03:
The install instructions accidentally referred to the pigpio lib as 'pigpiod' and thus apt-get fails to find and install the package. Instead the instructions should be:

`sudo apg-get install pigpio`

iFrostizz, 2020-06-04:
Hello, here is what I get:
```
sudo apt-get install pigpiod
[sudo] password for pi:
Reading package lists... Done
Building dependency tree
Reading state information... Done
pigpiod is already the newest version (1.71-0~rpt1).
0 upgraded, 0 newly installed, 0 to remove and 30 not upgraded.
```

What's your error please ?
Maybe you should try a `sudo apt update` according to @jneilliii

mcstoufer, 2020-06-07:
It could be that I'm running an older Octoprint distro (Raspbian GNU/Linux 9 (stretch) OctoPrint version : 1.4.0 OctoPi version : 0.16.0),  and the package name changed. Do you have another repo to pull packages from?
Here is what I'm seeing:
```
sudo apt-get install pigpiod
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package pigpiod
```
From my install:
```
 sudo apt-get install pigpio
Reading package lists... Done
Building dependency tree
Reading state information... Done
pigpio is already the newest version (1.64-1).
0 upgraded, 0 newly installed, 0 to remove and 138 not upgraded.
```
Also, now that I've got it all together with a separate 5v power supply ( standard raspberry pi 
 5v 2.5a power supply) driving the servos, It appears I'm getting nothing but noise on those two pins. The octoprint.log is showing successful commands to change the values of the two pins, but the servos just sit there thrashing around. I ended up getting 5v Towerpro servomotors ([1528-1076-ND](https://www.digikey.com/product-detail/en/adafruit-industries-llc/169/1528-1076-ND/5154651)) from DigiKey. Was there a specific servo model intended to be used here?

mcstoufer, 2020-06-08:
Closing out this issue. After moving up to a newer version of Octopi(0.17.0) and doing all the work, I found out that I had a floating voltage between the pins and the external power supply. By tying a raspberry pi ground pin to the same ground as the external ps, all the thrashing is gone. The movement now is acceptable, but I do get a funny bounce after horizontal rotation sometimes. There must be some feedback in the system that is keeping the servo off target. I'll figure that out.

Opening a new issue to update the wiring description to show the recommended double ground wiring.

closed 2020-06-08 by mcstoufer

iFrostizz, 2020-06-08:
This bounce is caused by the weight and the small power of these cute little servos :)
But I'm updating the code so the movement will not be longer instant
Big thanks for updates !