mledan/OctoPrint-EasyServo #8: Update wiring diagram to show common ground between external PS and Pi

mcstoufer, 2020-06-08:
I was having a lot of noise on the two pins before I tied a pi ground pin to the same breadboard row that the ground from the external PS was tied to. Once done, no more noise and servos only moved on command.

Requesting that the diagram be updated, or amended to inform the user that a common ground is recommended.

iFrostizz, 2020-06-08:
Thanks for this, Yep grounding is very important in electronics, I'm updating now!

iFrostizz, 2020-06-08:
In my case, I could wire the servos on the pi directly. Maybe you could try this (if your power supply is available)

iFrostizz, 2020-06-08:
Please see the update below the diagram

closed 2020-06-08 by iFrostizz

mcstoufer, 2020-06-08:
> In my case, I could wire the servos on the pi directly. Maybe you could try this (if your power supply is available)

I tried that first, but OctoPrint complained it was seeing a lot of undervolt warnings when driving those two servos.