mledan/OctoPrint-EasyServo #11: Just an idea

Fas0ne, 2020-07-02:
Hello, sorry if this is not the right place for this but an idea has been floating in my big empty head for weeks now, but unfortunately I lack the skills and knowledge to make it work.
Ok so basically , is there a way to remote gpio to another Pi. I am using a single Pi to run a couple of instances of Octo so my Pi is not In/on the printer and running wires just for auto on/off and servos is just messy for me. I do have a coupe of Pi Zeros and I thought " Hey! Why dont I just control that stuff directly from the Zero so there will be no more wires outside of the printer except for the usb." , But then I remembered that I have no actual idea of how to write code so... I'm here asking smarter people to do it.

iFrostizz, 2020-07-03:
Hey!
I like your idea, but it's not really the aim of my plugin. I think that if you want to do that, you should check for bluetooth modules, or connect your pi Zero in ssh and make them talk with your Octoprint raspberry. You want to do that for your enclosure, right ?
Well... You'll need only 1 wire to connect to your raspberry (the signal) per servo, it'll save you a lot of headhaches :)
Good evening ;)

Fas0ne, 2020-07-03:
Hey right back, thank you for the response. Yes I did think of some RF/ Bluetooth solution, even maybe an ESP module , but that's additional hardware and I think I have everything I need in the Pi already. Yes it's for my enclosed printer , but I think the unenclosed one would appreciate it if there was less clutter :smile: . I am definitely going to play around with this for a while, I did find this : https://gpiozero.readthedocs.io/en/stable/remote_gpio.html
and I think I can do something with it if I start changing some of your plugin's code, or after a few days I'll realize that this is not my thing and just buy a couple of Pi 3/4s and just stick one in each printer. The issue can be closed now, but I will let you know if I ever get this working the way I want it, maybe someone will be interested.

iFrostizz, 2020-07-04:
I would love to see that working ! :)
You could fork my plugin

iFrostizz, 2020-07-04:
@Fas0ne I think that you should replace the function that sends data to the servos in the Easyservo plugin by something that sends data on your Raspberry zero, you should work on it :)

Fas0ne, 2020-07-09:
@iFrostizz Well I got it "working" but still am not happy with it. The function( ) was a no go, because I wanted to keep the full functionality of the script. In the end the way to get it working was frustratingly simple(and it only took me a week to see it), I just changed this `self.pi = pigpio.pi()` to this `self.pi = pigpio.pi(10.0.0.21)` (my PiZero's local IP) in __init__.py and the servo on the PiZeroW started working. The problem now is that I can not get `__init__.py` to get an IP value from  `settings.plugins.EasyServo` `EasyServo_settings.jinja2` , which for me is important since I need to configure this for two instances of Octoprint on a single Pi to two separate PiZeros. Any idea how can I implement this in to `__init__.py` ?

iFrostizz, 2020-07-09:
Check how the plugin is made to (for example) change the GPIOX pin.
There is a default value that is stored in the memory of the plugin, and this value can be changed in the `settings.jinja2` This is all you need to do this. What you can also use, is in the `init.py, line 84`  `_setting.set` to write it in the config.yaml of the plugin, and to retreive the value, use `_settings.get` as in the `line 71`
Is it what you asked ?

Fas0ne, 2020-07-09:
Thanks , I will try again tomorrow. Just one more thing I dont get , and sorry if this is a stupid question( which it most likely is) - `self._logger.info` , what is it how does it work? I looked for a log but I couldn't find one related to what goes on within the plugin.

>

iFrostizz, 2020-07-10:
I guess that I commented theses logs. I use them for debugging and see how to plugin is working when there is a problem to see the value of each variable.
You can use it while writing `self._logger.info(hello)` and then you can see the logs in `octoprint.logs` in `Octoprint>logs>octoprint.log`