mledan/OctoPrint-EasyServo #12: Can this work with a Pimoroni Pan/Tilt hat over i2c?

frekel, 2020-11-08:
Hi, 

I really would love to pan/tilt my camera. I have a Pimoroni Pan/Tilt hat over i2c.

Can this plugin also handle this pan/tilt hat?

Thanks,
Frank

https://pinout.xyz/pinout/pan_tilt_hat

iFrostizz, 2020-11-18:
Hello,

The plugin should work fine on your board how it is already built.
The library used is called "pigpio" and make possible the PWM control after installing it, which means that you could control as much servos as possible as they are GPIO pins. (later update?)
Yours only seems to control servos on the I2C pins, so only 2 servos at maximum.
Please follow the procedure that is described in the README.md [https://github.com/iFrostizz/OctoPrint-EasyServo/blob/master/README.md](url)

iFrostizz, 2020-11-19:
So to conclude, no it'll not work using the Pimoroni Pan/Tilt hat with his library.
But using the pigpio library (Easy Servo plugin), you'll be able to control your servos without even needing any external board, just with your RPi. If you still wanna use it, you'll have to edit the code
Sorry for the imprecise last answer 😄

frekel, 2020-11-23:
Thanks for the info.  ;)

So just to be sure: If I get the Pimoroni Pan/Tilt working with the pigpio, I can (with some changes to your code) use this?

Or are you saying I can't control the hat with pigpio?

iFrostizz, 2020-11-23:
You ***cannot*** use the Pimoroni Pan/Tilt using the pigpio library. If you really want to use it for some reason, you'll have to remove the pigpio library from the python code, and replace all the related functions using the syntax provided by the library.
But you can, using the plugin (no change needed) and without the Pimoroni Pan/Tilt board, control your servos. Give it a try !

frekel, 2020-11-23:
I don’t have any servo’s to test with and I don’t want to disconnect my servo’s from the hat. 

I will close this issue as “resolved” 

Thanks again for your replies!!

closed 2020-11-23 by frekel

iFrostizz, 2020-11-23:
> I don’t have any servo’s to test with and I don’t want to disconnect my servo’s from the hat.
> 
> I will close this issue as “resolved”
> 
> Thanks again for your replies!!

No problem.
Do you think that it would be worth to make a cloned version of this plugin but adapted to the hat ? Do you see an interest ?

frekel, 2020-11-23:
I would definitely use it 😜

My time is limited to code it myself, but I would happily test any code someone needs tested!

iFrostizz, 2020-12-02:
Hi,

I just updated the plugin. Now, you can choose between the pigpio and the pimonori library. I can't test it so could you try it for me quickly ? 
Also, now you'll be able to invert the pan and tilt axis since you cannot choose the GPIO pins as for the pigpio library.
Good evening !
Also, you can add me on Discord if you have any feedback, maybe it would be more convenient to talk

[OctoPrint-EasyServo-master.zip](https://github.com/iFrostizz/OctoPrint-EasyServo/files/5631538/OctoPrint-EasyServo-master.zip)

frekel, 2020-12-02:
Hi,

I am currently not at my house. Will check it the Friday. Keep you posted!

(Sorry, no discord here)

frekel, 2020-12-03:
@iFrostizz What's you discord name?

cavoknosig, 2021-05-18:
I'm also trying to use the plugin with the pimoroni option / pantilt via i2c.
I can control the pantilt head via dedicated python script, but it doesn't move from within octopi / your plugin?!
Any idea what the cause could be? Also, the "get position" option doesn't seem to return anything. Nor do the home buttons trigger any action.

frekel, 2021-05-19:
@cavoknosig I will spin up my machine this afternoon and report back what version I have so we can debug your problem some more. 

The author helped me a lot and I did have it working like a charm!

keep you posted!

Frank

iFrostizz, 2021-05-19:
@cavoknosig Did you restarted your Octoprint server after changing the library used ? If yes, the pimoroni library should be displayed at the bottom of the settings.
I pushed an update recently that maybe broke the compatibility with pimoroni, do you have any clues in the Octoprint logs ?

cavoknosig, 2021-05-19:
thanks for the hint with the octopi logs!
it seems there is an issue with a dependency... respectively a typo in the plugins install/setup?! but there was no hint whatsoever about any error or missing dependency in the frontend? maybe that can be added to the plugin install?
i did by the way install all the pimoroni stuff via all the various "official" pimoroni variants, so it's unlikely there was anything missing from there (plus the pantilt already worked directly from scripts)

```
2021-05-19` 12:27:57,816 - octoprint.plugin - ERROR - Error while calling plugin EasyServo
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/pantilthat/pantilt.py", line 71, in setup
    from smbus import SMBus
ModuleNotFoundError: No module named 'smbus'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/__init__.py", line 271, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/EasyServo/__init__.py", line 129, in on_after_startup
    pantilthat.pan(self.angle_to_pimoroni(xAutoAngle))
  File "/home/pi/oprint/lib/python3.7/site-packages/pantilthat/pantilt.py", line 466, in servo_one
    self.setup()
  File "/home/pi/oprint/lib/python3.7/site-packages/pantilthat/pantilt.py", line 77, in setup
    raise ImportError("This library requires python3-smbus\nInstall with: sudo apt-get install python3-smbus")
ImportError: This library requires python3-smbus
Install with: sudo apt-get install python3-smbus

```

The problem is, when I try the apt-get install, it gives me: 
python3-smbus is already the newest version (4.1-1).

any ideas?

cavoknosig, 2021-05-19:
tried alternatively
sudo pip install smbus
sudo pip3 install smbus
it definitely installs and is present...
maybe some uppercase/lowercase or similar issue?

frekel, 2021-05-19:
Just to test:

can you try:
`I located the file /usr/lib/python3/dist-packages/smbus.cpython-35m-arm-linux-gnueabihf.so and copied it to 
/usr/lib/python3/dist-packages/smbus.so `

cavoknosig, 2021-05-19:
in the mentioned directory, i only have these two libraries:
smbus.cpython-36m-arm-linux-gnueabihf.so
smbus.cpython-37m-arm-linux-gnueabihf.so
-35m is not there.

so copied the latest 37m to smbus.so (did not exist before) and rebooted. no change, same error.

cavoknosig, 2021-05-19:
after lots of hacking and trying to follow whatever i found in forums, i finally did find the solution:
it seems to be related to virtual environments. unfortunately, apt-get etc... from the terminal does not directly solve the problem.
so assuming the installation of 
`sudo apt-get install python3-smbus`
has succeeded,

i then had to copy 

`/usr/lib/python3/dist-packages/smbus.cpython-37m-arm-linux-gnueabihf.so`
to
`/home/pi/oprint/lib/python3.7/site-packages/smbus.cpython-37m-arm-linux-gnueabihf.so`

and it finally worked. i hope that can somehow be automated from the plugin installer?!