mledan/OctoPrint-EasyServo #13: Set home coordinates?

andykaplan, 2020-11-15:
Awesome plugin.  Built a pan-tilt servo rig, seems to work GREAT.

Only issue:  Home is in the "wrong" place.  The camera is tilted back too far and twisted to the right.

Any way to set this position?

iFrostizz, 2020-11-18:
If there is no strange behaviour from the servos, you should place the servos at their middle (90° in angle or 1500us in pulse width) and then insert the axis. The aim here is to have this angle when your pan-tilt is in his neutral position. Of course, you can do wathever you want regarding your needs.

andykaplan, 2020-11-18:
Thanks.  I’ll need to disassemble and rebuild.

(Was hoping for a way to move the servos around using your arrow buttons, then click “set” to make that the home position — but I get that’s a bunch of work.)

iFrostizz, 2020-11-18:
That could be a dream if our cheap servos could turn continuously, and this feature could be added but it is not the case. What is the name of your servo ?
For the "cheap" and most common ones (SG90), the servo will only be able to go from 0 to 180° (~500 --> ~2500us) and if the pulsewidth that was set is out of the boundaries, it could simply break the servo rotor by forcing on it. Not good...

andykaplan, 2020-11-18:
Makes sense — although I don’t think most people would need more than 90 degrees, so there is some room on either side to spare…

(Yes, cheap and common servos, SG90)

I’ll just take my mount apart and re-do the servo rotor position.

iFrostizz, 2020-11-19:
Good luck!

closed 2020-11-19 by iFrostizz