mledan/OctoPrint-EasyServo #15: Updating from v0.1.1 to v0.2.1 fails

evilnoxx, 2020-11-25:
Hello,
Today I had a notification in my octoprint that there was an update for easy-servo. I clicked update, let the update run, and everything seemed to work fine. However, after reloading octoprint, the notification was still there, and nothing about the plugin had changed.

Looking at the updater output it seems it reinstalls v0.1.1 for some reason instead of updating.

Here is the output:

> Updating, please wait.
+++++++++++++++++++++++++++++++++
Now updating Easy Servo to v0.2.1
+++++++++++++++++++++++++++++++++
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install https://github.com/iFrostizz/OctoPrint-EasyServo/archive/v0.2.1.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting https://github.com/iFrostizz/OctoPrint-EasyServo/archive/v0.2.1.zip
Downloading https://github.com/iFrostizz/OctoPrint-EasyServo/archive/v0.2.1.zip (975 kB)
Requirement already satisfied (use --upgrade to upgrade): Easy-Servo==0.1.1 from https://github.com/iFrostizz/OctoPrint-EasyServo/archive/v0.2.1.zip in /home/pi/oprint/lib/python3.7/site-packages
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from Easy-Servo==0.1.1) (1.4.2)
Requirement already satisfied: pigpio in /home/pi/oprint/lib/python3.7/site-packages (from Easy-Servo==0.1.1) (1.78)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.12.1)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (7.1.2)
Requirement already satisfied: markupsafe<2.0,>=1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.1.1)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.5.0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2.0)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.18.2)
Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2020.8.7)
Requirement already satisfied: PyYAML<6,>=5.3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (5.3.1)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.18.0)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2020.9.23)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.0.7)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.4)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2.8.5)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (3.1.1)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2020.9.27)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.1.5.post0)
Requirement already satisfied: netifaces<1,>=0.10.9 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.10.9)
Requirement already satisfied: netaddr<1,>=0.7.19 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.8.0)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.0.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (5.1.1)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.57.0)
Requirement already satisfied: psutil<6,>=5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (5.7.2)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.2.0)
Requirement already satisfied: watchdog<1,>=0.10.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.10.3)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.1.0)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (3.4)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.2)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (4.0)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.6.0)
Requirement already satisfied: Jinja2<3,>=2.11.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2.11.2)
Requirement already satisfied: feedparser<6,>=5.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (5.2.1)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (2.24.0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.1.1)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (0.4.21)
Requirement already satisfied: flask<2,>=1.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.1.2)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.0.1)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->Easy-Servo==0.1.1) (1.5.0.1)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->Easy-Servo==0.1.1) (2.0)
Requirement already satisfied: urllib3>=1.10.0 in /home/pi/oprint/lib/python3.7/site-packages (from sentry-sdk<1,>=0.15.1->OctoPrint->Easy-Servo==0.1.1) (1.25.10)
Requirement already satisfied: certifi in /home/pi/oprint/lib/python3.7/site-packages (from sentry-sdk<1,>=0.15.1->OctoPrint->Easy-Servo==0.1.1) (2020.6.20)
Requirement already satisfied: setuptools>=36 in /home/pi/oprint/lib/python3.7/site-packages (from markdown<3.2,>=3.1->OctoPrint->Easy-Servo==0.1.1) (50.3.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Easy-Servo==0.1.1) (2020.1)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->Easy-Servo==0.1.1) (2.8.0)
Requirement already satisfied: six in /home/pi/oprint/lib/python3.7/site-packages (from websocket-client<1,>=0.57->OctoPrint->Easy-Servo==0.1.1) (1.15.0)
Requirement already satisfied: pathtools>=0.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from watchdog<1,>=0.10.2->OctoPrint->Easy-Servo==0.1.1) (0.1.2)
Requirement already satisfied: pyasn1>=0.1.3 in /home/pi/oprint/lib/python3.7/site-packages (from rsa==4.0->OctoPrint->Easy-Servo==0.1.1) (0.4.8)
Requirement already satisfied: chardet<4,>=3.0.2 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Easy-Servo==0.1.1) (3.0.4)
Requirement already satisfied: idna<3,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.23.0->OctoPrint->Easy-Servo==0.1.1) (2.10)
Building wheels for collected packages: Easy-Servo
Building wheel for Easy-Servo (setup.py): started
[... cut ...]
Building wheel for Easy-Servo (setup.py): finished with status 'done'
Created wheel for Easy-Servo: filename=Easy_Servo-0.1.1-py3-none-any.whl size=5536 sha256=ac25edeeca8775f0a0d8c6ed3c26028a62866a2c6fe21417a03e8c25ba29b706
Stored in directory: /tmp/pip-ephem-wheel-cache-pgn_rokm/wheels/de/b1/1f/bb94a8fbed6cdcf8931601446c48eb38adc240bd48efc664d0
Successfully built Easy-Servo
Installing collected packages: Easy-Servo
Successfully installed Easy-Servo-0.1.1
The update finished successfully and the server will now be restarted.

iFrostizz, 2020-11-25:
Hello,

Yes I forgot to replace a file that set the plugin version. I'm currently fixing some more issues on this version and improving it also, so I'll make sure not to forget this file with it in the next release (in some few days).
Sorry for that!

TodWulff, 2020-11-27:
If you do a manual install, putting this url in the proper field, it should update to 0.2.1 and stop the notifications:

https://github.com/iFrostizz/OctoPrint-EasyServo/archive/master.zip

At least it did for me.

iFrostizz, 2021-02-01:
Hello, it should be fixed with the new release.

closed 2021-02-02 by iFrostizz