mledan/OctoPrint-EasyServo #16: Feature Request: Add configuration for xy-offset of home postion / Control Servos with disconnected printer

der-den, 2021-01-03:
Hi,

my camera shows in the "HOME" direction not on the bed. So i would very nice i can setup a offset for the home position.

And, if possible, i like to use the camera for surveillance the full of my room (buzzwords: "child and cats"). But to control the servos i must turn on the printer. Can you change this?

Thanks.

iFrostizz, 2021-01-03:
Hi!

There is already a "xAutoAngle" and the same for the y axis in the plugin settings. This is set to 90° by default, but you can change it for sure.
I'll make the buttons usable even if the printer is turned off in the next update

der-den, 2021-01-04:
xAutoAngle: Ohh, who can read has a clear advantage
printer is turned off: perfect, thanks!

closed 2021-01-04 by der-den