mledan/OctoPrint-EasyServo #17: Feature request?

mebillica, 2021-01-03:
Hello, thank you for this. 
    I'm trying to create a camera slider for my 3d printer. 
    I would like to move it with this servo plug in. But the 90° limit holds me back. I have servos that can move 360° unlimited range. 
    Maybe a way to "reset" current position to its home? Then I could step it 1°, call that new home,  step it 1°, call that new home. Over and over as long as I want. 
    Then I could figure my total possible movement steps, divide that by my print z steps, and have it move those degrees on every layer change.

Thoughts?

Bill
[image]

iFrostizz, 2021-01-05:
I've never used unlimited range servos but I could adapt the plugin for this. Do you have the reference of these servos so that I can investigate a little bit ?
Also, your idea of Z axis tracking is really good, it could be done using simple trigonometry knowing the distance between the servo vertical axis and the middle of the print bed, we could deduce the angle in fonction of the current print height.

iFrostizz, 2021-01-05:
And this is why we would use trigonometry, the servo angle difference for each z hop isn't linear because of the distance between the servo and the print itself.

mebillica, 2021-01-13:
The unlimited range is a mod you do to the servo. Basically you cut off its endpoint in the gear, and it just keeps going. Here is a link. 

https://youtu.be/XUm2nj7te8E

What I was thinking is to use the servo, like a stepper motor. Figure out the distance of my camera track by moving the servo 1° at a time, from start to finish. Let's say it takes 692, 1° "steps". Then if I have a print that is 32 layers high. I divide the 692 possible 1° movements by the 32 layers to get 21. So every Z layer change, tell the servo to move 21° and get an "orbital" camera effect as it also would take a picture on the z layer change. It could be different for everyone too, and very customizable if people get creative. Right now  in my picture above, the track I am designing is a 12" radius from the center of the X build location on my MK3.

mebillica, 2021-01-13:
You could get some interesting shots I think. One I thought of that would be interesting: have the servo move to the left and right end points,  every other z hop, while moving inwards to the center. So using my numbers above, the snapshots would be at 0, then 692, 21, 671, 42, 650  63, 629, etc... I think it would be a blurry effect that would slowly focus? Not sure how it would look. But might be interesting.

iFrostizz, 2021-01-17:
Although the continuous rotation mod is interesting, I'm not sure that it would work very well. By opening the loop of the servo, you basically remove the angling reading and thus, you'll never know when asking to rotate at a certain speed for a certain amount of time if the servo rotated the correct angle. But I'll investigate on this, to see if there is a way to get some correct calculation.
The issue is that the DC motor that is driving the servo gears is different regarding your servo reference and may vary between users.

mebillica, 2021-01-17:
Hmmm. Yeah that sorta makes sense. But definitely above my knowledge. My thought was to send a signal to move  +/-  1° (or whatever value). Even with different servos, I thought it would work. Just that it would be relative to each servo. I know with my rc transmitter, it just keeps going due to no hard stop.

iFrostizz, 2021-01-21:
Hey !
I just got an idea.
So the user could choose which servo is continuous and could control it using a kind of joystick (as for your rc) that would go back to the neutral position when not used after having set the neutral position (in micro seconds). That way, you could be able to control the servo with more or less speed in directions. But I'm not sure to implement an angular correction because it would not be reliable at all. Or the only way would be to make a calibration wizard so that sending a certain value for a certain amount of time would make rotate the servo the desired angle but this is not trivial.
Also, don't you have to replace (or add?) the internal potentiometer of the servo by a resistor to "fix" the value that is read by the controller and thus, to fix this value as the perfect middle ?

mebillica, 2021-01-24:
I sorta follow what you're saying. Not sure how many people would want to control the servo with a joystick. Personally I'd prefer it to be automated. Also, to use the rc transmitter, you would need a receiver plugged in to receive commands from the transmitter. As far as replacing or adding the potentiometer, I don't think so. All there is, is a plastic tab on a gear inside. When that tab hits its end point, it stops. You just cut that tab off, and put it all back together. Then it spins continuously.

iFrostizz, 2021-01-24:
I was meaning a joystick on the control tab of Octoprint, that would come back to the middle point if not held.
In my understanding, the control pcb of the servo is reading the potentiometer value in order to achieve the correct rotation using the DC motor. That way, it would be better to have the potentiometer just at the middle to be able to control the biggest range around this middle value.
For instance, if you are sending a pulsewidth of 1500us (that should be the center for a normal servo and no rotation for a continuous servo), and that the potentiometer is not at its middle point, the servo may turn. (A classical servo range is 500-2500 micro seconds)
If this middlepoint is set let's say at 1000us (because this potentiometer is not centered), the 1000us command will be the one where the servo wouldn't turn, the 500us is the one for which the servo would turn in a given direction at a certain speed, and the 2500us command would make it turn in the opposite direction and much faster. But I could be wrong, and this is why I need to try this mod on one of my servos. It'll be fun :)

mebillica, 2021-01-24:
Ah, ok. Again I kinda follow all that. But definitely above my electronics and programming knowledge. Which is next to zero, lol.

iFrostizz, 2021-01-25:
Ahaha, no problem
I'm handling this, please hold on

iFrostizz, 2021-02-01:
Hello,
I'll post very soon an update of the plugin but it will not contain the continuous servo rotation support sadly. I still need to work on it.
By the way, are you using a small classic 9g continous servo for your slider ?

mebillica, 2021-02-01:
Ok. Yes, I use the 9g servo for my dollar tree foam board planes. Flite Test sells them.

https://store.flitetest.com/flite-test-es08aii-9g-analog-servo-flt-3032/p785288

Bill

mebillica, 2021-02-01:
Oops, what did I do? I'm new to github. Wanted to reply to your comment, but it "closed" this??

iFrostizz, 2021-02-01:
It's pretty impressive, isn't is the servo having some trouble in order to move all this mass ?

mebillica, 2021-02-01:
To be honest, I haven't hooked it all up yet. Was kinda waiting to see what kind of control I could do fitst. I just drew up and printed the track and mount. But it shouldn't be a problem. It's really light weight, and I can slide it all easily by hand.

Zeleni-Zub, 2021-05-12:
Hi, I know it's been long since last post here but I just wanted to let you know that there is a continuous rotating servo, the FS90R. Physically same as 180deg SG90 9G. In case you are tinkering still with this and regular SG servo misbehaves maybe FS one will work better :) Also both servos have 1,5kg torque, pretty impressive for such small things!
iFrostizz thank you so much for the plugin, works like a charm! Before I installed it I had to keep a running script in terminal window to input servo angles if I wanted to move it. 
A bit offtopic in this thread but would it be simple/possible to restrict movement so you can't go out of those bounds by using the buttons or set angle window in Octoprint? Meaning, in my case the tilt angles should be restricted to 40-180deg because angle lower than 40 would start to break the cam mount.

iFrostizz, 2021-05-12:
Hey!
To be honest, I was wondering if I should close this issue or not because I didn't got any consistent results with the SG90 9G servo, but I'm gonna try the FS90R for sure! I'm really glad that this plugin helps you.
And sure, this is possible ! I'm gonna push an update in 2 days maximum.
Good day!

Zeleni-Zub, 2021-05-12:
Wow, you're awesome! Thank you!

iFrostizz, 2021-05-14:
Hi Zeleni-Zub,
Could you try this version please ?
Sadly, I'm not at home for some time and cannot test any hardware...
[OctoPrint-EasyServo-master.zip](https://github.com/iFrostizz/OctoPrint-EasyServo/files/6477403/OctoPrint-EasyServo-master.zip)
Hope it's gonna work, and I'm gonna update the plugin then :)

Zeleni-Zub, 2021-05-14:
Hi iFrostizz,

Awesome work my friend!!
Works without an issue, thank you so much!
This tweak will save a lot of cam mounts hehehe

Thanks again!

iFrostizz, 2021-05-14:
Great!!
I'm gonna add some securities like not being able to have a minimum angle greater than maximum angle and push an update.

iFrostizz, 2021-05-18:
So @mebillica , sorry  for the delay. I ordered some FS90R continuous servos and gonna play with them in more than one month when I'll be back home.

iFrostizz, 2021-06-12:
@mebillica 
So I just received the FS90R continuous servos and it works well, the four arrows are enough to control them. The only thing that I could add is to set the middle of the pulse width (about 1500us but could change) in the settings, and remove all the absolute angling stuff.
Does it works on your side ?

mledan, 2023-02-01:
Closing this as it was released some time ago. I have also tested it here with continuous servos

closed 2023-02-01 by mledan