mledan/OctoPrint-EasyServo #20: Raspi-config forgot

Bubu38, 2021-06-07:
hi, nice plugin , really good job, you just forgot in raspi-config to enable network gpio communication.
I had an error everytime i sudo service pigpiod start...
Thanks to you ! bye

iFrostizz, 2021-06-08:
Hi, are you talking about the "remote GPIO" option ? It should not be needed as it is used to control GPIO on another pi that is connected to a network https://bennuttall.com/whats-new-gpio-zero-v1-3/
It seems like this feature is starting the pigpio daemon, which is needed to complete further steps. Maybe that you forgot that one ?
Thanks !

Bubu38, 2021-06-08:
Ok but how do you enable pigpio daemon otherly on a fresh new octopi?

iFrostizz, 2021-06-08:
It is explained in the "pre installation requirements"

Bubu38, 2021-06-08:
I did enable I2C ARM in raspiconfig but didn't work like that. I had to enable gpio remote to enable daemon

iFrostizz, 2021-06-08:
This thread could be interesting for you to start the daemon and then check if it's effective : https://www.raspberrypi.org/forums/viewtopic.php?t=199323

closed 2021-06-09 by iFrostizz