mledan/OctoPrint-EasyServo #21: EasyServo not running within octoprint-docker environment

ertraid, 2021-07-22:
Running Octoprint-docker successfully with other plugins (e.g. GPIO Control, PSU Control), but getting errors with Easy Servo. Pardon my very limited knowledge, but specifically, seems the plugin does not recognize/ access pigpiod that is already running on the host.. 
Understand docker might not be in scope for your plugin, and Ok with that, but thought I'd report int anyway in case you want to tackle.
See this thread for detail: 
https://community.octoprint.org/t/easyservo-plugin-not-working-for-me-any-ideas/35770/7

Let me know if I can provide further info.

Regards, ET

dkmode, 2023-03-08:
Quick and dirty solution is to add in https://github.com/mledan/OctoPrint-EasyServo/blob/master/EasyServo/__init__.py#L116

`self.pi = pigpio.pi(HOSTNAME)` 
hostname of your Raspberry Pi.
Just Docker's localhost vs host's localhost. :)

Ideally this address could be exposed as just another variable from EasyServo plugin settings.

mledan, 2023-03-14:
Hey @dkmode thanks for commenting! Can you explain what this achieves in terms of this issue?

dkmode, 2023-03-14:
> Hey @dkmode thanks for commenting! Can you explain what this achieves in terms of this issue?

Just wanted to keep it short, cause it's minor problem.

Full context:
Octoprint running in docker on Raspberry Pi 4, installed EasyServo plugin, configured pgpiod on host, connected servos to gpio pins etc.
1. Servos works outside of container via pgpiod daemon (listening locally on 0.0.0.0 when service edited with `-x -1` opts)
2. By default `self.pi = pigpio.pi()` connects to `localhost`, (or some local socket, one solution is to run pgpiod in same container as Octoprint, but that just creates more problems)
3. With `self.pi = pigpio.pi(HOSTNAME)` in my case eg. `rpi4.lan` or with another hostname which can be reached from inside of octoprint container (eg. when running pgpiod in compose).
4. My suggestion is to expose pgpiod hostname variable (with default localhost) in settings UI and init pgpiod client with that hostname.

Thanks for work on plugin. :)

mledan, 2023-03-22:
@dkmode Got a clean install on my pi to finally test this fix; I'll be taking a look this week and early next week to see if I can get it working from within a container

Here's a pre-release with the change you described if you'd like to test it as well:
https://github.com/mledan/OctoPrint-EasyServo/archive/refs/tags/v0.2.5.zip

@ertraid I would love to hear your feedback as well if this resolves the issue for you. 

Let me know if either of you need help with installing from the link. Thanks!

ertraid, 2023-04-07:
Hi again... Firstly, thanks both for your support.

@mledan, I uninstalled my older plugin and then installed v0,2,5 using your url above but noticed no change in behaviour. One strange thing is that the plugin still gets reported as 024 in plugin manager, and also during the installation process:

> _Installing plugin from https://github.com/mledan/OctoPrint-EasyServo/archive/refs/tags/v0.2.5.zip...
> /usr/local/bin/python -m pip --disable-pip-version-check install file:///tmp/tmp7lh7cmtw/OctoPrint-EasyServo-0.2.5.zip --no-cache-dir
> Processing /tmp/tmp7lh7cmtw/OctoPrint-EasyServo-0.2.5.zip
> Preparing metadata (setup.py): started
> Preparing metadata (setup.py): finished with status 'done'
> Requirement already satisfied: OctoPrint in /usr/local/lib/python3.8/site-packages (from Easy-Servo==0.2.4) (1.8.7)
> Requirement already satisfied: pigpio in /octoprint/plugins/lib/python3.8/site-packages (from Easy-Servo==0.2.4) (1.78)
> [...many more Requiremetns already satisfied...]
> Building wheels for collected packages: Easy-Servo
> Building wheel for Easy-Servo (setup.py): started
> Building wheel for Easy-Servo (setup.py): finished with status 'done'
> Created wheel for Easy-Servo: filename=Easy_Servo-0.2.4-py3-none-any.whl size=11815 sha256=cb284db0adcfe015282279778af9f066c7648656ad16eb134290fb3df973bdbc
> Stored in directory: /tmp/pip-ephem-wheel-cache-95fgqr68/wheels/e9/3b/e1/19ac51535f188341f57a0ee8286e96cc1a4e8ead52b02c0dbc
> Successfully built Easy-Servo
> Installing collected packages: Easy-Servo
> Successfully installed Easy-Servo-0.2.4
> Done!_

Not sure if this is correct (i.e. not all updates are made in this pre-release) or the link is pointing to the incorrect code?
At any rate, no solution so far... I'll keep an eye for any further updates.

Thanks again, 
ET

ertraid, 2023-04-07:
Took a further look at my octoprint.log.. thought they might be of use (let me know if you want the complete file).

Thsi is teh latest run (with 025):
> 2023-04-07 17:34:42,766 - octoprint.plugins.EasyServo - INFO - The libraryUsed is pigpio
> 2023-04-07 17:34:42,767 - octoprint.plugin - ERROR - Error while calling plugin EasyServo
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 273, in call_plugin
>     result = getattr(plugin, method)(*args, **kwargs)
>   File "/usr/local/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
>     return f(*args, **kwargs)
>   File "/octoprint/plugins/lib/python3.8/site-packages/EasyServo/__init__.py", line 115, in on_after_startup
>     self._logger.info("Initializing pigpio with hostname: {}".format(HOSTNAME))
> NameError: name 'HOSTNAME' is not defined

This is an older run (using 024):
> 2023-04-07 16:59:09,899 - octoprint.plugins.EasyServo - INFO - The libraryUsed is pigpio
> 2023-04-07 16:59:09,900 - octoprint.plugins.EasyServo - INFO - Initializing pigpio
> 2023-04-07 16:59:09,905 - octoprint.plugins.EasyServo - INFO - <pipio.pi host=localhost port=8888>
> 2023-04-07 16:59:09,905 - octoprint.plugins.EasyServo - INFO - False
> 2023-04-07 16:59:09,909 - octoprint.plugin - ERROR - Error while calling plugin EasyServo
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 273, in call_plugin
>     result = getattr(plugin, method)(*args, **kwargs)
>   File "/usr/local/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
>     return f(*args, **kwargs)
>   File "/octoprint/plugins/lib/python3.8/site-packages/EasyServo/__init__.py", line 127, in on_after_startup
>     self.pi.set_servo_pulsewidth(GPIOX, self.angle_to_width(xAutoAngle))
>   File "/octoprint/plugins/lib/python3.8/site-packages/pigpio.py", line 1678, in set_servo_pulsewidth
>     return _u2i(_pigpio_command(
>   File "/octoprint/plugins/lib/python3.8/site-packages/pigpio.py", line 1025, in _pigpio_command
>     sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
> AttributeError: 'NoneType' object has no attribute 'send'

Hope it helps. Let me know if you want the full log file.
ET

mledan, 2023-04-18:
> 

Hey thanks for your replies and logs. I recreated the pre-release and it seems to have taken care of the versioning problem. https://github.com/mledan/OctoPrint-EasyServo/releases/tag/v0.2.5-alpha

Will be adding some additional logging around it this week to try and pin down the hostname issue

ertraid, 2023-04-29:
Tks @mledan ...tried the alpha right now, but still getting a similar error:

> 2023-04-29 13:33:36,119 - octoprint.plugins.EasyServo - INFO - The libraryUsed is pigpio
> 2023-04-29 13:33:36,119 - octoprint.plugin - ERROR - Error while calling plugin EasyServo
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.8/site-packages/octoprint/plugin/__init__.py", line 273, in call_plugin
>     result = getattr(plugin, method)(*args, **kwargs)
>   File "/usr/local/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1688, in wrapper
>     return f(*args, **kwargs)
>   File "/octoprint/plugins/lib/python3.8/site-packages/EasyServo/__init__.py", line 115, in on_after_startup
>     self._logger.info("Initializing pigpio with hostname: {}".format(HOSTNAME))
> NameError: name 'HOSTNAME' is not defined

Hope it helps...
ET