mledan/OctoPrint-EasyServo #22: Problem with end stop

jpb69400, 2021-09-19:
Hello, thank you for this plugin.

My hardware.
Raspberry Pi 4 Model B Rev 1.4, Python version 3.7.3, Octoprint version 1.6.1

My Easy Servo Settings (list of change)
	Y Axis inversion
	X Minimum Angle: 40
	Y Minimum Angle: 40
	X Maximum Angle: 140
	Y Maximum Angle: 140
	
My test
	X = 40° ==> OK
	X = 140° ==> KO
	
	Y = 40° ==> KO
	Y = 140° ==> OK
	
When i move the servo to end value (for x= 140° or for y = 40°) after, it is impossible to move the axis after.
The only way to solve the problem is a reboot system.
The problem is the same with or without servos connected.

Regards, JPB

iFrostizz, 2021-09-19:
Hello,

That could be an issue with the precision of the pulse width because of cheap servo.
There is a log written when a servo hit a boundary saying roughly `GPIO ... reached his boundaries with a ... pulse width`.
Could you tell me the value of the pulse width in this case please ?

Thanks!

jpb69400, 2021-09-19:
Thank you for your quick response.
I do not think that the quality of the servos is in question, the 
problem is the same without connected servo (I monitor the position of 
the axes on the "Easy Servo Settings" screen).
2021-09-19 21:03:27,743 - octoprint.plugins.EasyServo - INFO - GPIO 12 
reached his boundaries with a 943 pulse width
2021-09-19 21:04:00,867 - octoprint.plugins.EasyServo - INFO - GPIO 13 
reached his boundaries with a 2056 pulse width
Kind regards

jemkein, 2021-10-19:
Hi!
I got the same Problem, any Fixes?

iFrostizz, 2021-10-21:
Hey @jpb69400,
Indeed, the angles were inverted, I pushed an update and it seems to work properly now.
Good idea! I will add a checkbox to choose wether or not appending the "Get Current Position" button with the current coordinates in the control tab tomorrow hopefully.

BecoKo, 2022-01-08:
Hi! Here's a simple fix on this issue. You may be want to replace:
`for x in range(actual_width + 1, width_to_reach + 1, incrementSign):`
from master branch with:
`for x in range(actual_width + incrementSign, width_to_reach + incrementSign, incrementSign):`
to get proper boundaries from range function.

Regards

iFrostizz, 2022-01-08:
Hi @BecoKo , thanks for your contribution !
The fix was already pushed in the devel branch as discussed above, but the issue is still open since it's not merged yet, I'm quite busy these days ... Thanks again

closed 2023-01-30 by mledan

mledan, 2023-01-30:
@BecoKo Released the above mentioned changes. Please let me know if you have any issues going forward. Thanks!