mledan/OctoPrint-EasyServo #23: third servo

gortador, 2022-10-05:
Hello,
I use your plugin which I find great, however I use a third servo to control a third position possibility to move the camera on a y axis. So a third servo. Is it possible to add an option to your plugin to order an additional servo?

Thanks a lot

iFrostizz, 2022-10-05:
Hey,
Thanks! And no, using more than 2 servos cannot be done at the moment, it was planned to be able to add as much servos as we want but I'm not currently maintaining the plugin.
Have a nice day!

gortador, 2022-10-05:
> Hey, Thanks! And no, using more than 2 servos cannot be done at the moment, it was planned to be able to add as much servos as we want but I'm not currently maintaining the plugin. Have a nice day!

ok, too bad, thank you very much for answering!

iFrostizz, 2022-10-05:
> > Hey, Thanks! And no, using more than 2 servos cannot be done at the moment, it was planned to be able to add as much servos as we want but I'm not currently maintaining the plugin. Have a nice day!
> 
> ok, too bad, thank you very much for answering! and there would not be to your knowledge a plugin that will allow me to control a servo directly from octoprint

Sorry,
Not to my knowledge: https://plugins.octoprint.org/search/?q=servo

mledan, 2022-12-22:
@gortador what kind of servo control hardware are you using, if any?

gortador, 2022-12-22:
hello, i am using two sg90 servos

gortador, 2022-12-22:
> @gortador what kind of servo control hardware are you using, if any?

hello, i am using two sg90 servos

gortador, 2022-12-22:
> @gortador what kind of servo control hardware are you using, if any?

@mledan :  hello, i am using two sg90 servos

mledan, 2022-12-22:
> two sg90 servos
👍 

Are they connected directly to your pi or are you using something like: https://www.sparkfun.com/products/15316
to control the servos

mledan, 2022-12-22:
@gortador I'm working on adding support for up to 16 servos via the sparkfun board ^^, I'll let you know when complete

gortador, 2022-12-23:
> @gortador I'm working on adding support for up to 16 servos via the sparkfun board ^^, I'll let you know when complete

@mledan no, directly 😊

oldmanbluntz, 2023-04-05:
Have you seen this plugin? It may work for you.

https://plugins.octoprint.org/plugins/robotcontrol/

oldmanbluntz, 2023-04-05:
> @gortador I'm working on adding support for up to 16 servos via the sparkfun board ^^, I'll let you know when complete

What if you wanted 196 servos?

Just kidding around, but i'm also working on something like this, but based on the PCA9685 from adafruit. 12 bit 16 channel I2C PWM controller, with the max capibility of running 196 servos in total when you hook up the max 62 PCA9685's you can daisychain.

gortador, 2023-04-05:
Hello ,

I am looking to use 5 servos, but if I can use more but on the i2c port of the raspberry that is not enough I would not like to use an arduino type card... thank you for your news!

Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

mledan, 2023-04-07:
> What if you wanted 196 servos?
>
> Just kidding around, but i'm also working on something like this, but based on the PCA9685 from adafruit. 12 bit 16 channel I2C PWM controller, with the max [...]

Love it! Would absolutely love to support an arbitrary amount of servos (imagine the UI/UX for that 😃), I'll grab that board this summer. I am currently refactoring the plugin to support the strategy pattern which should enable setting up individual motors on specific libraries and yes, ideally will not care about a max number of motors or even pi hats (eg. for purposes of testing all this; I'm setting my pi up to use both the pan/tilt with the pimoroni library +pi-hat and hooking into unused gpio pins  to control additional motors to wipe my hot-end (pun not intended)

gortador, 2023-04-07:
no problem for puns, I am French-speaking and do not master the subtleties of English.  however, if I can give you some suggestions regarding the end use, it would be really useful to be able to name each servo separately as well as possibly choose a default angle of movement, per pulse, or even choose the maximum number of angles  and min and to start the movement if you can too, make sure that we can use an additional endstop to stop the race, as well as the speed of movement... but maybe I'm asking a lot ^^ I  will be happy to do the beta tester if you would like my help or advice 😀
-------- Message d'origine --------De : mledan ***@***.***> Date : 07.04.23  07:06  (GMT+01:00) À : mledan/OctoPrint-EasyServo ***@***.***> Cc : gortador ***@***.***>, Mention ***@***.***> Objet : Re: [mledan/OctoPrint-EasyServo] third servo (Issue #23) 

@gortador I'm working on adding support for up to 16 servos via the sparkfun board ^^, I'll let you know when complete

What if you wanted 196 servos?
Just kidding around, but i'm also working on something like this, but based on the PCA9685 from adafruit. 12 bit 16 channel I2C PWM controller, with the max capibility of running 196 servos in total when you hook up the max 62 PCA9685's you can daisychain.

Love it! Would absolutely love to support an arbitrary amount of servos (imagine the UI/UX for that 😃), I'll grab that board this summer. I am currently refactoring the plugin to support the strategy pattern which should enable setting up individual motors on specific libraries and yes, ideally will not care about a max number of motors or even pi hats (eg. for purposes of testing all this; I'm setting my pi up to use both the pan/tilt with the pimoroni library +pi-hat and hooking into unused gpio pins  to control additional motors to wipe my hot-end (pun not intended)

mledan, 2023-04-10:
All great suggestions! I will definitely take note to consider these as I'm updating the settings UI to accommodate a larger number of motors.